kornia/kornia-rs #1: Fix dockerfile

edgarriba, 2022-03-07:
The dockerfile doesn't work well and we want to Integrate the ci with our docker container

carlosb1, 2022-03-08:
A prototype was done here https://github.com/carlosb1/bindgen-dlpack-rs, we should copy and link this project in a offical repository...

carlosb1, 2022-03-08:
I had a problem with a libturbo dependency, the rust library uses debian and it uses an old version, this docker worked for  me.
`❯ cat devel.Dockerfile 
FROM rust:alpine

RUN apk add --update --no-cache python3 && ln -sf python3 /usr/bin/python
RUN apk add --update --no-cache libjpeg-turbo-dev libjpeg-turbo py3-pip sudo git musl-dev libc-dev libressl-dev mesa-dev

RUN python3 -m pip install --upgrade pip setuptools setuptools_rust

WORKDIR app
COPY . .
RUN pip3 install .

`
We need to move the context to get sourc code, but it works.

edgarriba, 2022-03-12:
fixed in #4

closed 2022-03-12 by edgarriba