kornia/kornia-rs #3: Automate the packaging for manywheels in the ci.yml

edgarriba, 2022-03-07:


carlosb1, 2022-03-08:
First step could be the creation of the ci.yml file and check that it can run.. for the packaging, it implies to include a cargo publish with the crates.io credentials...

carlosb1, 2022-03-28:
Some example about how can we do this?

carlosb1, 2022-05-08:
I am not sure but it seems that this task was done, I check the release.yml and it has the ' messense/maturin-action@v1' git action for applying manywheels.

edgarriba, 2022-05-09:
It's not activated, there were some issues with the docker environment. Right now I was doing the releases by hand with support only for Linux. Would be great if you can explore that using the dockerfile for release and automate every time we publish a release that builds and push the wheels to pip

closed 2022-07-05 by edgarriba