kornia/kornia-rs #6: Support properly OpenEXR

edgarriba, 2022-03-14:
the crate `image-rs` supports openexr images so that we can import/export floating point images. However, the `cv::Tensor` class assumes for now only byte representation.

carlosb1, 2022-05-08:
https://github.com/cessen/openexr-rs/blob/master/examples/simple_input.rs - can be done something like this to import images to tensor?

edgarriba, 2022-05-09:
With the current image-rs we already can decode exr, however we need to somehow redesign the cv:: Tensor class so that can hold not only uint8 but also floating point data.

gau-nernst, 2023-02-27:
I think using Rust generics might work? I can try when I have some time.

edgarriba, 2025-03-09:
@gau-nernst are you still interested in that ?

viktorho, 2025-03-29:
Hi @edgarriba ,
Has this issue already been resolved or not? I tried using a tensor with f32, and it worked well. Could you please clarify this issue further? I would love to help

edgarriba, 2025-03-29:
Yes, that’s probably;y the way to go to return an Image<f32,3>

closed 2025-12-28 by edgarriba