kornia/kornia-rs #37: No MacOS ARM64 wheels for Python 3.10 and 3.11

gau-nernst, 2024-02-10:
As shown here https://pypi.org/project/kornia-rs/0.0.8/#files

Also, a separate issue. On my Mac M1, when I run `pip install kornia-rs`, it tries to install 0.0.1 version from source (and fails). This is probably because only 0.0.1 version has source distribution (https://pypi.org/project/kornia-rs/0.0.1/#files, see the .tar.gz file), while newer versions don't have. When I run `pip install git+https://github.com/kornia/kornia-rs`, it can be built from source just fine.

edgarriba, 2024-02-11:
That’s possibly a flag in here: https://github.com/kornia/kornia-rs/blob/main/.github/workflows/release.yml#L36
Beware about the project refactor though https://github.com/kornia/kornia-rs/pull/32

gau-nernst, 2024-03-14:
Yup, works for me now.

On a separate note, we probably should run tests for kornia-rs on various OS x Python combinations? I notice that currently only Ubuntu runs tests now. I think now GitHub Actions have free MacOS ARM64 runner for open source projects?

closed 2024-03-14 by gau-nernst

edgarriba, 2024-03-15:
added here: https://github.com/kornia/kornia-rs/pull/52/files#diff-057cc96278abef7805c36e31271c952e066432c7be2c98b1b13e368ba3c557a8R34