kornia/kornia-rs #39: not use `anyhow` to handle errors

edgarriba, 2024-03-09:
for now errors are handle with `anyhow` -- this is good at application level but ideally  we should create custom error types for lib development

vcode11, 2024-04-16:
Hi @edgarriba I am interested in contributing to this project. How can I join the community to get started?

edgarriba, 2024-04-16:
@vcode11 thanks for the interest ! if you feel comfortable with Rust and `thiserror` feel free to go for it

vcode11, 2024-04-17:
I found your commit: https://github.com/kornia/kornia-rs/commit/d72fca813731e6359201e4f35bce9ffb1fb621df this seems to solve the problem? Any particular reason why you didn't merge this? Anything that is missing here that you would like to have?

edgarriba, 2024-04-20:
lack of time to finally land this but feel free to take that diff resume the task

edgarriba, 2024-09-03:
implemented in main already

closed 2024-09-03 by edgarriba