kornia/kornia-rs #46: implement pyramidal optical flow lk

edgarriba, 2024-03-12:
reference codes:
- https://github.com/kornia/kornia/pull/2658
- https://docs.opencv.org/4.x/dc/d6b/group__video__track.html#ga473e4b886d0bcc6b65831eb88ed93323

requirements:
- gaussian pyramid: https://github.com/kornia/kornia/blob/main/kornia/geometry/transform/pyramid.py
- gaussian blur https://github.com/kornia/kornia/blob/main/kornia/filters/gaussian.py#L13
- convolution filter https://github.com/kornia/kornia/blob/main/kornia/filters/filter.py#L38

thedgarg31, 2025-03-16:
Implementation of Pyramidal Lucas-Kanade Optical Flow (#46)

I have implemented the Pyramidal Lucas-Kanade (LK) Optical Flow as discussed in this issue. The key changes include:

Gaussian Pyramid Construction: Utilized existing Kornia functionality to build image pyramids.
Multi-scale Optical Flow Estimation: Integrated the LK method at multiple scales to enhance tracking accuracy.
Gaussian Blur and Convolution Filtering: Leveraged Kornia's built-in filters to preprocess images.
Test Cases: Added a unit test (optical_flow_test.rs) to validate functionality.
Please review the implementation and let me know if any modifications are needed. Looking forward to your feedback!

cjpurackal, 2026-01-14:
Re-opening as part of efforts to implement SLAM modules

namanguptagit, 2026-02-24:
OK i am working on it will raise the pr asap

edgarriba, 2026-10-04:
Implemented: `calc_optical_flow_pyr_lk` in `crates/kornia-imgproc/src/optical_flow_pyr_lk.rs` (#755). Verified on main @ be0ac0c. Closing — reopen if something is missing.

closed 2026-10-04 by edgarriba