kornia/kornia-rs #47: implement `warp_affine` / `warp_perspective`

edgarriba, 2024-03-12:
reference:
- warp_perspective: https://github.com/kornia/kornia/blob/main/kornia/geometry/transform/imgwarp.py#L48
- warp_affine: https://github.com/kornia/kornia/blob/main/kornia/geometry/transform/imgwarp.py#L131

needs:
- grid_sample / map_coordinates

edgarriba, 2024-03-28:
feel free to go for it -- about grid_sample, it will basically a way to reuse code. If you see in the `[resize_native](https://github.com/kornia/kornia-rs/blob/main/src/resize.rs#L159)` function the `ndarray::Zip` is basically parallelizing over the destination points. In any case you will have to reimplement that to make it efficient, we don't want regular looping but instead iterators (which is what's happening in that code already.

gau-nernst, 2024-03-28:
Interesting. Do you have benchmarks between naive for-loop version and the current version?

Some observations
- Currently we materialize the whole index array (width x height). This is quite wasteful.
- There is probably overhead for parallelization (from what I see, ndarray uses rayon https://docs.rs/ndarray/latest/ndarray/parallel/index.html). Perhaps we can parallelize over the rows only instead of all the points to reduce overhead.

In any case, let's get a working implementation first before looking more into the speed.

edgarriba, 2024-06-17:
done

closed 2024-06-17 by edgarriba