kornia/kornia-rs #51: No Module Kornia_Rs

Dacrikka, 2024-03-14:
come on guys, I don't even know if this is the right place to write. I've been using Stable Diffusion XL on Windows for a year and have never had any problems. Yet today I can't start anything because there is a warning that says "no module kornia_rs" and everything stops. What should I do? 
I tried to delete Venv folder...but nothing. Any advice? I'm completely stuck.

edgarriba, 2024-03-14:
have you tried with `pip install kornia_rs` ? in principle the latest kornia release should automatically download the package for you

Dacrikka, 2024-03-15:
> have you tried with `pip install kornia_rs` ? in principle the latest kornia release should automatically download the package for you

Yes, I did. But nothing... I reinstalled SD from zero, and it works. And the two folders (old one and new one) are the SAME....sorcery

edgarriba, 2024-03-15:
good to hear that it worked ! we made a release this week, which makes kornia-rs a hard requirement in kornia. But as far as i see stable diffusion relies on  fixed version. Thanks for reporting this, we'll keep in mind.

closed 2024-03-15 by edgarriba