kornia/kornia-rs #57: fast image resize

edgarriba, 2024-03-17:
Explore this crate and benchmark: https://docs.rs/fast_image_resize/latest/fast_image_resize/index.html

ternaus, 2024-03-18:
```
Darwin Vladimirs-MacBook-Pro.local 23.2.0 Darwin Kernel Version 23.2.0: Wed Nov 15 21:53:18 PST 2023; root:xnu-10002.61.3~2/RELEASE_ARM64_T6000 arm64
```

```
kornia_rs==0.1.2
numpy==1.26.4
opencv-python-headless==4.9.0.80
pillow==10.2.0
```

```
python py-kornia/benchmark/bench_resize.py
OpenCV: 0.04 ms
PIL: 0.20 ms
Kornia: 0.17 ms
```

I believe that the benchmark should run over many different images. 

This setup could be a bit too extensive, but still....
https://github.com/ternaus/imread_benchmark/blob/main/imread_benchmark/benchmark.py

edgarriba, 2024-03-19:
I think this benchmarks should be running in a Linux machine which I guess is the most typical scenario for training. Besides, why don’t you use timeit which run several iterations and remove any possible outlier plus other things for you. Also, the benchmark should really be done against a data loader routine and measure probably IPS (images per second)

ternaus, 2024-03-19:
Just run your code from https://github.com/kornia/kornia-rs/blob/main/py-kornia/benchmark/bench_resize.py

Would be happy to check a version with an updated setup for a benchmark :)

edgarriba, 2024-06-17:
> Explore this crate and benchmark: https://docs.rs/fast_image_resize/latest/fast_image_resize/index.html

DONE

closed 2024-06-17 by edgarriba

ternaus, 2024-06-18:
Do you have comparison with other libraries, somthing similar to:

https://github.com/albumentations-team/albucore/blob/main/benchmark/results/uint8_3.md

and if resize in kornia-rs is not broken, as it is in torchvision, tensorflow or OpenCV it would be ultra great.https://github.com/assafshocher/ResizeRight

edgarriba, 2024-06-27:
@ternaus i do not have such benchmark. Feel free to contribute -- i'm currently focusing on feature complete for the Rust side since is what i have to use more recently at work to prototype stuff to validate things before implementing in Cuda/VPI, so i do not look too much on the performance for now, just creating baselines.

For the resize, i believe that kornia/pytorch has this implemented as expected https://x.com/kornia_foss/status/1388149222697054212 -- i haven't tried with the rust version. For the next kornia-rs versions i'm working on a new Tensor class to replace ndarray, so i might spend some time refactor some low level kernels for geometric transforms.