kornia/kornia-rs #81: which one should I use, kornia-rs or opencv-rs for my use case?

Raj2032, 2024-06-16:
I want to design a real time application where a live video camera will detect certain images on billboards. I want to use  [yolox](https://yolox.readthedocs.io/en/latest/train_custom_data.html?highlight=fine%20tune#tips-for-best-training-results) and create my own custom models.

For my use case which one is more suited, opencv-rs or kornia-rs?

edgarriba, 2024-06-16:
Thanks for the interest! I think that for your application you could give a shot to the [webcam example](https://github.com/kornia/kornia-rs/blob/main/examples/webcam/src/main.rs) and combine with the [ort](https://github.com/pykeio/ort) crate in case there’s an ONNX artifact of the model that you want to deploy. Happy to help with any implementation.

Raj2032, 2024-06-17:
Hey mate, thanks for your message :)

Just wondering since I am using yolox, is ONNX like some format for yolox?

johnnv1, 2024-07-03:
> Hey mate, thanks for your message :) Just wondering since I am using yolox, is ONNX like some format for yolox?

ONNX is a "common" (production cases) format in which we usually export the models. read more at https://onnx.ai/onnx/intro/

So to use the ONNX will depend on the yolox implementation you'll use.

Raj2032, 2024-07-03:
Will do thank you mate :)

edgarriba, 2024-09-02:
closing this for now. Please, reopen if needed

closed 2024-09-02 by edgarriba