kornia/kornia-rs #124: Add apriltag detector support

edgarriba, 2024-09-04:
Integrate [`apriltag`](https://docs.rs/apriltag/latest/apriltag/) crate which is useful for robotics and camera calibration tasks.

Probably put into a new `kornia-calib` crate.

powei-lin, 2024-11-16:
Hi, I removed those three functions in my crate. It's' more robust now.

powei-lin, 2024-12-22:
I added the support for kornia in aprilgrid in 0.5.0.
https://github.com/powei-lin/aprilgrid-rs/blob/cb4e56ee4bd1a24a074ada3d97571e21e351c6b3/tests/test_detector.rs#L37

MuhammadBilalKhan267, 2025-03-14:
Hello, I would like to contribute to this.

Aren't most already implemented and just need to be ported? More information on what exactly to implement and port would be appreciated.

edgarriba, 2025-12-28:
Supported natively

closed 2025-12-28 by edgarriba