kornia/kornia-rs #126: Interop `kornia::Tensor` with `ort::Tensor` (including cuda)

edgarriba, 2024-09-05:
Integration with CUDA buffers would have to go through a different API, [`TensorRefMut::from_raw`](https://docs.rs/ort/2.0.0-rc.5/ort/type.TensorRefMut.html#method.from_raw). It just needs a pointer and shape, so Kornia's existing APIs suffice. `TensorRefMut`s can be used zero-copy in session inputs just like a `Value`, but since the memory is user-managed, it does have a lifetime bound. See ort's `cudarc` example: https://github.com/pykeio/ort/blob/60f6ecae9064a75a474678998af42ea87067b1b1/examples/cudarc/src/main.rs#L35-L43

_Originally posted by @decahedron1 in https://github.com/kornia/kornia-rs/pull/125#discussion_r1745777792_

kylekam, 2025-03-26:
Is there a reason why [kornia tensor requires an array type for shape and stride](https://docs.rs/kornia-rs/latest/kornia_rs/tensor/struct.Tensor.html)? This restricts the use to known shapes at compile time, which makes dynamic conversion between ORT tensors and kornia tensors challenging. [ORT returns a slice/vec](https://docs.rs/ort/2.0.0-rc.9/ort/value/struct.Value.html#tymethod.try_extract_raw_string_tensor) for the tensor shapes.

Code sample.
```
  // read an image
  let data: ort::value::Tensor<u8>;
  {
      let image: Image<u8, 3> = F::read_image_any_rgb8(&args.image_path)?;
      let ort_tensor = ort::value::Tensor::from_array((image.shape, image.into_vec()))?;
      data = ort_tensor;
  }

  // assume data is passed to another function
  // try to convert ORT tensor back to kornia tensor
  let (shape, data_vec) = data.extract_raw_tensor_mut();
  let data_vec = data_vec.to_vec(); // returns Vec<u8>
  let kornia_tensor = Tensor::from_shape_vec(
      shape,
      data_vec,
      CpuAllocator,
  );
```
has the following compile error
```
error[E0308]: mismatched types
   --> main.rs:64:9
   --> main.rs:64:9
    |
63  |     let kornia_tensor = Tensor::from_shape_vec(
    |                         ---------------------- arguments to this function are incorrect
64  |         shape,
    |         ^^^^^ expected `[usize; _]`, found `&[i64]`
    |
```

edgarriba, 2025-03-26:
The tensor design is meant to be type safe at compile time to avoid runtime bugs. A way to adress this ticket could be to extend tensor with a `Tensor::unsqueeze` method (or possibly a trait) that can be especialized by especific tensor ranks

kylekam, 2025-03-26:
>The tensor design is meant to be type safe at compile time to avoid runtime bugs. A way to adress this ticket could be to extend tensor with a Tensor::unsqueeze method (or possibly a trait) that can be especialized by especific tensor ranks

Without replacing shape as [usize;N], I don't see how a trait could be used to dynamically define the field during runtime.

My understanding is that if we want shape to support [usize;N], it would still require the type to be generic in some way.
```
  trait DynShape {
      fn get_shape(&self) -> Vec<usize>;
  }
  
  impl<const N: usize> DynShape for [usize; N] {
      fn get_shape(&self) -> Vec<usize> {
          self.to_vec()
      }
  }
  
  impl DynShape for Vec<usize> {
      fn get_shape(&self) -> Vec<usize> {
          self.clone()
      }
  }
  
  struct Tensor {
      shape: Box<dyn DynShape>,
  }

  let array_shape: Box<dyn DynShape> = Box::new([1, 2, 3]);
  let vec_shape: Box<dyn DynShape> = Box::new(vec![1, 2, 3]);

  let array_tensor = Tensor{ shape: array_shape };
  let vec_tensor = Tensor{ shape: array_shape };
```

edgarriba, 2025-03-27:
We can just have specialised unsqueeze2, unsqueeze3, etc. That tries to put into the new rank. I would like to avoid dynamic strict a for now and keep things relatively simple.l without having to complicate much the apis

kylekam, 2025-03-29:
Should GPU mem allocation with kornia tensors be handled by cubecl? I see only CpuAllocator behavior is implemented right now. This work may be related: https://github.com/kornia/kornia-rs/issues/250