kornia/kornia-rs #143: Tensor::get not failing for index out of bounds in last dimension

jandremarais, 2024-09-21:
I would expect that index [0, 2] of a 2x2 matrix is out of bounds but the following assert fails:

```rust
let data: Vec<u8> = vec![1, 2, 3, 4];
let t = Tensor::<u8, 2>::from_shape_vec([2, 2], data, CpuAllocator)?;
assert!(t.get([0, 2]).is_none());
```

Since `t.get_iter_offset([0,2])` returns `2` and `2` < `4` (size of data buffer), buffet.get does not fail.

edgarriba, 2024-09-22:
Can you extend test cases and attempt to fix that function ?

Something in this direction should work 
```rust
for (i, &dim_size) in self.shape.iter().rev().enumerate() {
        let idx = index[N - 1 - i];  // Reverse the index to match dimensions
        if idx >= dim_size {
            return None;  // Out of bounds
        }
        offset += idx * stride;
        stride *= dim_size;  // Update stride for next dimension
    }
    ```

jandremarais, 2024-09-22:
with #144

closed 2024-09-22 by jandremarais