kornia/kornia-rs #149: Support reading images other thant RGB

edgarriba, 2024-09-26:
Expand `kornia_io::functional::read_image_any` to read / decode images in more formats such as u16

reference: https://github.com/kornia/kornia-rs/blob/main/crates/kornia-io/src/functional.rs#L104

NOTE: Take especial care when constructing the `Image` to avoid copies: https://github.com/kornia/kornia-rs/blob/main/crates/kornia-io/src/functional.rs#L130

edgarriba, 2024-09-26:
@emilmgeorge this is another use case for allowing `Tensor` / `Image` to be created with xero copy overhead

emilmgeorge, 2024-09-28:
> NOTE: Take especial care when constructing the `Image` to avoid copies: https://github.com/kornia/kornia-rs/blob/main/crates/kornia-io/src/functional.rs#L130

https://github.com/kornia/kornia-rs/blob/a4b5fd4b6084a48c3be5e3f97b7d568f48d0f53a/crates/kornia-io/src/functional.rs#L130

Here `img.into_rgb8().into_vec()` could be used instead:

- [`into_rgb8()`](https://docs.rs/image/latest/src/image/dynimage.rs.html#296-306) creates a copy only if `img` is not already in RGB8 format.
- [`into_vec()`](https://docs.rs/image/latest/src/image/buffer.rs.html#1325-1330) consumes the image and gives us the inner vector without copy. `to_vec`, on the other hand, seems to point to [`slice::to_vec`](https://doc.rust-lang.org/1.81.0/src/alloc/slice.rs.html#449-451) (according to rust-analyzer) which copies.

edgarriba, 2024-09-29:
@emilmgeorge awesome, I'll give it a shot. Also we can start supporting to return a generic `C`. What's your opinion to encode the PixelFormat too in the Image struct ? I like the current definition because is very flexible to have more generic imgproc operators, but at the same time i kind of miss the color space semantics.

edgarriba, 2024-09-29:
a small pov here https://github.com/kornia/kornia-rs/pull/154 feel free to comment

emilmgeorge, 2024-09-29:
Your implementation in #154 looks good. 👍🏼  I do not have any better ideas.