kornia/kornia-rs #167: Implement patch-tracker-rs

edgarriba, 2024-10-29:
- create a `kornia-tracking` module to implement `https://github.com/powei-lin/patch-tracker-rs`
- ideally not use nalgebra but [`glam`](https://github.com/bitshifter/glam-rs)
- needed things not available yet in kornia-imgproc:
  - [ ] `image_grad` https://github.com/powei-lin/patch-tracker-rs/blob/main/src/image_utilities.rs#L5
  - [ ] `build_image_pyramid` https://github.com/powei-lin/patch-tracker-rs/blob/main/src/tracker.rs#L145
  - [ ] `detect_keypoints` https://github.com/powei-lin/patch-tracker-rs/blob/main/src/image_utilities.rs#L90

aayushhzz, 2025-03-11:
Hey @edgarriba I have started implementing this patch-tracker-rs.

I just needed to ask about the structure of the module.

`https://github.com/powei-lin/patch-tracker-rs` should it be same like this one? or if you have anything specific you have in mind do let me know

edgarriba, 2025-03-13:
the link should be a reference implementation but open to adopt other ideas, like the LK sparse optical flow tracker. See also as ref: https://github.com/kornia/kornia/pull/2658

just as a comment, this is not a short project as it involves some many in between steps and proper benchmarkins

closed 2025-12-28 by edgarriba