kornia/kornia-rs #169: `find_countours`

edgarriba, 2024-11-05:


Udit-git-acc, 2025-04-01:
Is this issue resolved ?

edgarriba, 2025-04-01:
@Udit-git-acc not yet, but please focus on finishing first other opened PRs first

zeel991, 2026-02-25:
Hi @sidd-27 , I'd like to work on this issue, I've gone through the OpenCV docs and the Suzuki-Abe border following algorithm and all other functions they have used. Please assign this to me

I will raise the PR with all the functions and their tests and benchmarks

mathurojus, 2026-03-08:
Hey @sidd-27 , I’ve opened PR #797 to add find_contours with OpenCV‑style hierarchy modes to kornia-imgproc. When you get a chance, could you please review it and let me know if anything needs to be adjusted, or approve/merge it if it looks good?