kornia/kornia-rs #172: `image_grad` https://github.com/powei-lin/patch-tracker-rs/blob/main/src/image_utilities.rs#L5

edgarriba, 2024-11-05:


edgarriba, 2025-01-09:
maybe this should be implmented via separable convolution

Anishp-cell, 2025-12-23:
Hi! I’d like to work on this issue if it’s still available.
I’ll start by reviewing the Python Sobel reference and the patch-tracker Rust implementation to match Kornia’s style. Please let me know if there are any specific constraints or preferred API shape.

Anishp-cell, 2025-12-27:
Hi @edgarriba  
I’ve implemented `image_grad` using separable Sobel kernels in `kornia-imgproc`.
- Computes dx and dy directly
- Uses existing Sobel kernels + separable_filter
- Fully documented
- All tests pass
PR is ready for review. Happy to adjust if needed.

Nyx128, 2026-01-11:
Can someone tell me if there is any work left to be done here, as i would like to try and do it. I noticed the help wanted tag so please let me know.

edgarriba, 2026-10-04:
Implemented as `spatial_gradient_float` (plus parallel variants) in `crates/kornia-imgproc/src/filter/ops.rs`, alongside the sobel kernels. Verified on main @ be0ac0c. Closing — reopen if a differently-shaped `image_grad` API is still wanted.

closed 2026-10-04 by edgarriba