kornia/kornia-rs #226: Ship free-threaded (cp313t) wheels

ngoldbaum, 2025-01-22:
It looks like you've already updated to PyO3 0.23, so all that's left is adding CI on the free-threaded interpreter, adding testing for thread safety, and fixing any issues that are discovered in the process. 

There is some multithreaded testing and functionality on the rust side, but it looks like there isn't any testing of using `kornia-py` from the Python `threading` module.

I'm a PyO3 maintainer and helped lead the effort to ship support for free-threaded python in PyO3 0.23. I'm also actively working on ecosystem support for free-threaded Python as part of my job at quansight labs.

All that to say that I'm happy to do all the heavy lifting here, I'd just appreciate being able to ask questions and get code review.

edgarriba, 2025-01-23:
hi @ngoldbaum thanks for reaching out ! Would be awesome if you could lead that part. I've been recently focusing more on the Rust side of the project so any help to improve the Python bindings is very welcomed. For awareness, recently @johnnv1 have been helping a bit too here. Feel also free to join our discord to discuss more: https://discord.com/invite/HfnywwpBnD

edgarriba, 2025-03-07:
I believe this is done right ‘

ngoldbaum, 2025-03-07:
No this isn't done yet. I'm waiting on a PyO3 and rust-numpy 0.24 release before I can start working on kornia-rs.

ngoldbaum, 2025-03-24:
I updated the pyo3 and rust-numpy dependencies to 0.24.0 and it looks like the existing tests pass:

```
(kornia-py) goldbaum@goldbaum-QEMU-Virtual-Machine:~/kornia-rs/kornia-py$ PYTHON_GIL=0 pytest
============================================================= test session starts =============================================================
platform linux -- Python 3.13.2, pytest-8.3.5, pluggy-1.5.0
rootdir: /home/goldbaum/kornia-rs/kornia-py
configfile: pyproject.toml
collected 16 items                                                                                                                            

tests/test_color.py ...                                                                                                                 [ 18%]
tests/test_enhance.py .                                                                                                                 [ 25%]
tests/test_histogram.py .                                                                                                               [ 31%]
tests/test_icp.py ..                                                                                                                    [ 43%]
tests/test_io.py .....                                                                                                                  [ 75%]
tests/test_package.py .                                                                                                                 [ 81%]
tests/test_resize.py .                                                                                                                  [ 87%]
tests/test_warp.py ..                                                                                                                   [100%]

============================================================= 16 passed in 0.98s ==============================================================
(kornia-py) goldbaum@goldbaum-QEMU-Virtual-Machine:~/kornia-rs/kornia-py$ python -VV
Python 3.13.2 experimental free-threading build (main, Mar 17 2025, 20:49:10) [GCC 6.3.0 20170516]
```

Note that because of the torch dependency, I think right now this will only work on Linux unless you build pytorch from source. Also there's a `uv` free-threaded specific bug I ran into trying to adapt the justfile: https://github.com/astral-sh/uv/issues/12445, so that means free-threaded CI will probably be a special case for a little while.

@edgarriba are you ok with me sending in a PR that sets up free-threaded CI in its own actions workflow along with a TODO to integrate it better with the rest of the CI setup in the future?

The next step is to look at [`pytest-run-parallel`](https://github.com/Quansight-Labs/pytest-run-parallel) to hopefully shake out use of global state in the python bindings. I also plan to learn a little bit more about the library and think about object thread safety and multithreaded stress tests.

edgarriba, 2025-03-25:
Hi, thanks for the effort. Yes, please go ahead with the PR. About PyTorch, not sure I understand, as we are just downloading prebuilt wheels. One side topic, I’m gonna exploring soon the possibility to migrate the python bindings to full support pyarrow as with some discussion with other package maintainers. I also noticed in the past in some benchmarks that convert from/to numpy was adding quite a bit of overhead which seems that with arrow that won’t happen.

ngoldbaum, 2025-03-25:
> Yes, please go ahead with the PR.

Awesome, I'll work on that today.

> About PyTorch, not sure I understand, as we are just downloading prebuilt wheels.

PyTorch only ships free-threaded wheels for Linux right now, even on the nightly channel.

> One side topic, I’m gonna exploring soon the possibility to migrate the python bindings to full support pyarrow as with some discussion with other package maintainers. I also noticed in the past in [...]

Cool, definitely worth looking into IMO.

ngoldbaum, 2025-03-28:
I audited the python bindings for thread safety today and it looks like things are in pretty good shape, at least for `kornia-rs`. The `kornia` Python library is likely susceptible to race conditions in the mutable objects it exposes and you might want to think about documenting what (if any) thread safety guarantees you want to provide to protect against races and nondeterminism if someone tries to (ab)use threads.

I think the library as-is is "safe" in the sense that data races are impossible (due to the guarantees provided by CPython, PyO3, and rust).

In particular, it's nice to see what `PyImageDecoder` and `PyImageEncoder` ultimately wrap an `Arc<Mutex>` around the turbojpeg encoder and decoder, which are definitely not thread-safe.

The other exposed pyclasses are all immutable as well (I think you could add `frozen` to the pyclass declarations?), so they should also all be thread-safe.

I didn't look closely at the implementations of functions exposed in the bindings. Are any of them implemented using global state or have global side effects?

ngoldbaum, 2025-03-28:
After #320, the only `unsafe impl` left in the codebase is for `TensorStorage`:

```
crates/kornia-tensor/src/storage.rs
129:unsafe impl<T, A: TensorAllocator> Send for TensorStorage<T, A> {}
130:unsafe impl<T, A: TensorAllocator> Sync for TensorStorage<T, A> {}
```

This type wraps allocations for raw pointers, so there might be thread safety issues if `T` isn't Send or Sync, depending on the implementation of the allocator. It looks like the CPU allocator just wraps `std::alloc::alloc` and `std::alloc::dealloc`. I don't see any explicit documentation that e.g. [`GlobalAlloc::alloc`](https://doc.rust-lang.org/std/alloc/trait.GlobalAlloc.html#tymethod.alloc) is thread-safe, but I'd be pretty surprised if it wasn't.

One note, it does look like you're not checking for a non-zero sized `layout` in `CpuAllocator::alloc`, which the safety notes for `GenericAlloc::alloc` says could cause UB.

ngoldbaum, 2025-03-31:
> do you have any suggestion for it ? that's the only workaround I could find at that time

It's fine as-is, IMO.

That said, I'm not sure the `Arc` is actually necessary for the python bindings since the PyO3 smart pointers do [more or less the same thing](https://pyo3.rs/v0.24.0/class.html#bound-and-interior-mutability). 

You could do something like make e.g. `PyImageDecoder` wrap a Mutex<turbojpeg::Decompressor> and then rely on pyo3 to handle the `Send` and `Sync` issues. Using a `Mutex` rather than an `Arc<Mutex>` also allows using a frozen pyclass, e.g. https://github.com/PyO3/pyo3/issues/4980. This means you don't need to pay the cost of PyO3's runtime atomic reference counting either.

For the Rust API and whether or not the `Arc` makes sense there, I think the only real benefit of using an `Arc<Mutex>` over &Mutex is to allow creating static references that are shared between threads. With scoped threads you can often avoid the need to share static references like that.

> we can create some issues to tackle all this.

I don't think there are any thread safety issues in the Python API as-is. 

Making the pyclasses frozen would be a cleanup and performance optimization, but everything is already thread-safe IMO. I'm happy to open issues about moving to frozen pyclasses, since that's probably a performance improvement.

> Do we have any way to test ? would be great to find prove around the issues

If you're interested, we can add multithreaded stress tests. I think the best multithreaded stress tests are of the sort of parallel workflows you want to support. 

I don't think any of the pyclasses are particularly interesting to test. They're all immutable (at least from Python) or are safe because they use interior mutability behind a mutex. Also specifically for the encoder and decoder, Operating simultaneously on the same image encoder and decoder from multiple threads in Python is a strange thing to do. IMO writing tests to make sure that more or less amount to checking if pyo3 and the rust standard library mutex are thread-safe on a workflow that is not really supported is not the most useful to do IMO.

Are there any multithreaded workflows you imagine you might want to use an end-to-end multithreaded stress test in the test suite? Having pointers on what sort of multithreaded parallel workflows you'd like to support would be really helpful 😄 

That said, I don't expect to find anything but it'll certainly add piece of mind and help future refactors or provide code infrastructure for multithreaded testing if you want to expose mutable objects in the future.

ngoldbaum, 2025-04-02:
In the meantime I think I might send in a PR adding thread safety docs and declaring free-threaded support in the extension module. I don't think multithreaded stress tests are necessary to do that given that upstream guarantees would probably have to be violated to produce any thread safety issues.

That's the last step before updating the wheel-building infrastructure to build free-threaded wheels.

edgarriba, 2025-04-02:
@ngoldbaum opened this one https://github.com/kornia/kornia-rs/pull/334
couldn't make it work though with frozen classes

ngoldbaum, 2025-04-02:
cool, I’ll try messing around with that PR branch sometime in the next few days.

edgarriba, 2025-04-02:
awesome! feel free to push there

ngoldbaum, 2025-04-02:
not sure I can - want to give me a `kornia-rs` commit bit?

ngoldbaum, 2025-04-04:
I made #339

PR #347 (2025-04-07): Update python release automation, add free-threaded builds
Fixes #226 (🚀🧵)

Most of this is fixing bitrot in the Python release automation:

* Works around issues with cmake4 and turbojpeg-sys. I opened https://github.com/honzasp/rust-turbojpeg/issues/30 to track this upstream. Way more complicated than I originally expected because it turns out maturin-action might be running on any one of RHEL 7-9 or Ubuntu!
* Updates to upload-artifact and download-artifact v4, which required updating naming bookkeeping to avoid generating artifacts with duplicate names, which is now banned.
closed 2025-04-07 by edgarriba