kornia/kornia-rs #229: Speed compare with OpenCV

MonolithFoundation, 2025-01-25:
for instance, cvtColor, capture etc speed compare with OpenCV

edgarriba, 2025-01-26:
There’s no such benchmark as OpenCV is primarily written in c++ (and a lot of opencl) and the scope for now is to match or improve performance over native rust libraries.

edgarriba, 2025-01-26:
Said that, you are very welcomed to submit a patch with some benchmarks

edgarriba, 2025-03-09:
please, reopen if want to provide any fair updated benchmark

closed 2025-03-09 by edgarriba

MonolithFoundation, 2025-03-10:
@edgarriba  I think a benchmark comparsion could be more fair. However, if you guys didn't have time to do this or kornia-rs are in a very early stage, please move on for core abilities.

edgarriba, 2025-03-10:
The goal and focus of this library is to have state of the arte for rust native implementations. From our side almost each new feature is properly benchmarked against its rust native alternative if you check in the code base.

Anyone is more than welcomed to provide a benchmark against other non rust libraries.