kornia/kornia-rs #231: Implement native QRCodeDetector

edgarriba, 2025-02-05:
Create a native implementation of  QR detector similar to OpenCV [QRCodeDetector](https://docs.opencv.org/4.x/de/dc3/classcv_1_1QRCodeDetector.html) requested by user in discord.

Ideally as separated crate `kornia-qr` within the parent worksapce.

jaydon2020, 2025-02-28:
Hi, anyone working on this issue? I would like to work on this issue. 
@edgarriba cloud you provide me more detail and assign me to this issue.

start to create a new crate
```
cargo new kornia-qr --bin
```

About the detect and decode what data types should I return? any suggestion?
```
detectAndDecodeCurved(&self, input_image: Image<u8, 3>) -> retval, points, straight_qrcode
retval: String // Decoded data
points: Tensor<f32, 4> // box binding of the qr code found
straight_qrcode: Image<u8, 3> // rectifiedImage
```
References
https://github.com/opencv/opencv/blob/4.x/modules/objdetect/src/qrcode.cpp
https://github.com/dignifiedquire/quircs/tree/master

edgarriba, 2025-03-01:
Hi, sounds good to me if you want to give it a shot. What’s exactly retval as String ? Also for simplicity I would return for now a vec<[[f32; 2]; 4]> for the points. Are you planning to write from scratch, or use any other library ? Let’s minimise as much as possible the deps tree

pratheekhassan21, 2025-03-03:
Hi,@edgarriba is jaydon2020 working on this?Can i work on this issue

jaydon2020, 2025-03-03:
@edgarriba, I do some research this library working well on QRCode detect
- [quircs](https://github.com/dignifiedquire/quircs)
- [rqrr](https://github.com/WanzenBug/rqrr)

Since minimize as much as possible the deps tree, I will manually port it from OpenCV QRCode Detect.

Also, the OpenCV implement an abstract class GraphicalCodeDetector for QRCode, do we need to follow similar approach so that it provide an trait for future implementation like barcode or aruco detection

edgarriba, 2025-03-04:
@jaydon2020 sounds like a plan