kornia/kornia-rs #252: implement `write_image_png_gray8`

edgarriba, 2025-03-09:


Astik-2002, 2025-03-16:
Hello. I'm new to kornia-rs and i wanted to dive into the code. This seems like a good issue to start. Can you share some details as to how image writing needs to be done?

edgarriba, 2025-03-17:
> Hello. I'm new to kornia-rs and i wanted to dive into the code. This seems like a good issue to start. Can you share some details as to how image writing needs to be done?

already covered in #262

Astik-2002, 2025-03-17:
> > Hello. I'm new to kornia-rs and i wanted to dive into the code. This seems like a good issue to start. Can you share some details as to how image writing needs to be done?
> 
> already covered in [#262](https://github.com/kornia/kornia-rs/pull/262)

Oh great. Can you share if any other good first issues still need a solution? I'd be happy to contribute

edgarriba, 2025-03-18:
@Astik-2002 you can take a look at this one https://github.com/kornia/kornia-rs/issues/100

PR #332 (2025-04-02): Extend PNG encoding and decoding support
Fixes #252 
Fixes #253 

Also, I have added rgb16 and rgba16 support in both read and write functions. With this PR `png` module now supports every conversion.

I was thinking that `kornia_image::Image` should have a function for converting e.b. `Image<u16, 1>` to `Image<u8, 1>` and vice-versa, while maintain the same data using functions like [`u16::from_be_bytes`](https://doc.rust-lang.org/stable/std/primitive.u16.html#method.from_be_bytes)
closed 2025-04-02 by edgarriba