kornia/kornia-rs #276: Improve tensor traits so that we do not have to clone values

edgarriba, 2025-03-17:
Currently, `Tensor` / `TensorStorage` do not implement the `Clone` traits which means we have to explicitly clone variables every time we implement a logic on top of them.

Explore whether to implement the `Clone` / `Copy` traits in Tensor crate.

kylekam, 2025-03-19:
Anyone working on this? I can start looking into it.

GautamSharmaaa, 2025-03-21:
Hi, I am working on optimising `Tensor` and `TensorStorage` to eliminate unnecessary cloning while maintaining ownership safety and performance. My approach includes:  

1. Replacing unnecessary `.clone()` calls with borrowing (`&Tensor`).  
2. Using `Arc<Vec<T>>` for shared storage instead of cloning data.  
3. Implementing `Cow<T>` for copy-on-write optimisations.  
4. Adding in-place operations (`add_inplace`, `mul_inplace`, etc.) to avoid extra allocations.  
5. Introducing tensor views (`slice()`) to prevent redundant copies.  
6. Applying `RefCell<T>` or `Mutex<T>` where mutation is required without cloning.  
7. Benchmarking performance improvements to validate efficiency gains.  

This should significantly reduce memory overhead and improve execution speed. Let me know if you have any feedback or specific constraints to consider before I proceed with a PR.  

Thanks!

edgarriba, 2025-03-21:
@GautamSharmaaa sounds like a good proposal. Can you coordinate with @kylekam to avoid duplication work? I suggest to discuss all this is public discord, as I see so many changes that might need to happen iteratively in separated PRs

closed 2025-12-28 by edgarriba