kornia/kornia-rs #288: Replace `faer` by `kornia_linalg::svd3` in ICP crate

edgarriba, 2025-03-19:
Recently merged an optimized version for computing SVD for 3x3 matrices in https://github.com/kornia/kornia-rs/pull/279

The kornia_icp needs to get updates by the new `kornia_linalg::svd3` function.

ajit2704, 2025-03-22:
If no one is working, I like to start looking into it.

edgarriba, 2025-03-23:
@ajit2704 see https://github.com/kornia/kornia-rs/pull/299

IamRam3, 2025-03-23:
Hi @edgarriba , can you suggest me any issue. Because most of the issue's are either taken or Have some PR's linked to them for review.

prikshitsingh24, 2025-08-10:
@edgarriba I am working to fix this issue wanted to ask that the kornia-rs svd takes in glam::mat3 as input and returns glam::mat3 as the output while the existing icp is using fear mat everywhere for creating and performing operation wanted your suggestion on whether I should replace that with glam  or try to  convert the result back to fear compatible mat3.

prikshitsingh24, 2025-08-11:
I also wanted to ask that `svd` returns Mat3 of type f32 and also takes in Mat3 of type f32 but in icp we are working with f64 matrices 
should i reduce it to f32 or change svd to input and return f64 types