kornia/kornia-rs #319: kornia-io crates.io docs aren't building

ngoldbaum, 2025-03-28:
See https://docs.rs/crate/kornia-io/0.1.8/builds, last build failed on Jan 1 2025 due to a missing gstreamer install.

ankitmrj, 2025-03-28:
@ngoldbaum  I’d like to work on this issue. Please assign it to me.

ngoldbaum, 2025-03-28:
I’m not a maintainer

edgarriba, 2025-03-28:
@ngoldbaum thanks for pointing this out. I gave a try at some point but couldn't find out a proper way to setup the docs builder to use gstreamer properly. @ankitmrj do you have any intuition how to fix this ?

ankitmrj, 2025-04-01:
@edgarriba Thanks for the insight! I’ll investigate how to configure docs.rs to properly use GStreamer. One approach is checking whether it can be installed as a dependency via build.rs or .cargo/config.toml. If not, we could explore conditional compilation to disable GStreamer-dependent features during doc builds. I'll update soon!

AS1100K, 2025-04-01:
I briefly looked into gstreamer-rs repository and found that they are using an internal cfg flag named `docsrs` that disables the build script. I will open the PR soon, updating the `Cargo.toml`

PR #328 (2025-04-02): Fix `kornia-io` docs not building on docs.rs
Fixes #319 

The docs.rs fails to build the documentation because `gstreamer-app-sys` custom build script fails. [See Logs](https://docs.rs/crate/kornia-io/0.1.8/builds/1605403).

After inspecting the [`gstreamer-app-sys/Cargo.toml`](https://gitlab.freedesktop.org/gstreamer/gstreamer-rs/-/blob/main/gstreamer-app/sys/Cargo.toml#L63) and it's [build.rs](https://gitlab.freedesktop.org/gstreamer/gstreamer-rs/-/blob/main/gstreamer-app/sys/build.rs), I came to know that enabling `docsrs` flag would fix the issue. 

Also, `gio-sys` crate also uses `docsrs` [... cut]
closed 2025-04-02 by edgarriba