kornia/kornia-rs #333: Create Python API docs using Sphinx

ngoldbaum, 2025-04-02:
Right now the `kornia-rs` pypi package has a documentation link that points at https://kornia.readthedocs.io:

https://github.com/kornia/kornia-rs/blob/75220b4a1a3357c84ce711f500d45d4331809d2b/kornia-py/pyproject.toml#L35-L38

IMO this is suboptimal, it would be better if the kornia-rs python bindings had their own separate API docs that the Kornia docs can cross-reference where needed. This means you wouldn't need to update the kornia docs separately if, for example, you changed the API of `PyImageSize`: https://kornia.readthedocs.io/en/stable/image.html#kornia.image.ImageSize.

I realize it's a fair bit of sphinx work to do this; this is a long-term idea or an idea for someone who wants to work on documentation and build automation.

edgarriba, 2025-04-02:
That's a great suggestion !

SkaarFacee, 2025-06-03:
@edgarriba I would like to take up this issue  if no one has already started working on it

SkaarFacee, 2025-06-09:
@edgarriba Sorry for the delayed response. I got caught up with something else. I will work on it right away. Would it be okay if I discussed my tentative plan with you and make PR's as I go ?

SkaarFacee, 2025-06-13:
Hi @edgarriba @cjpurackal 
Thank you for the guidance. I've taken a look at the documentation for polars, rerun, and candle-py and have put together a tentative plan to tackle this issue. My goal is to create a modern, clean, and easily maintainable documentation site for the kornia-rs Python bindings.

I would appreciate your feedback on the following proposed plan. I intend to break down the work into smaller, incremental PRs for easier review and faster iteration.

Proposed Plan: Sphinx Documentation for kornia-rs

Phase 1: Project Initialization and Basic Setup

    Goal: Establish the foundational Sphinx documentation site.
    Actions:
        Create a new docs directory within the kornia-py subdirectory (or dedicated kornia-rs/docs repository, pending team decision).
        Initialize a new Sphinx project inside kornia-py/docs.
        Set up the basic conf.py with essential project information (project name, author, copyright, etc.).
        Add a simple index.rst as the main landing page.
        Update .gitignore to exclude the docs/_build directory.
    PR #1: feat(docs): initialize sphinx for python api
        This PR will include the initial Sphinx project structure and configuration.

Phase 2: Theme and Styling

    Goal: Select and implement a modern, readable theme that aligns with Kornia's brand.
    Actions:
        Add the chosen theme (e.g., furo - leaning towards this for its modern look, or consider the PyTorch theme for consistency with Kornia core) to the documentation dependencies.
        Configure conf.py to use the selected theme.
        Customize the theme with the Kornia logo and color scheme to maintain brand consistency.
        Self-correction/Improvement: Review how similar Rust-Python binding projects (e.g., Polars, Rerun, Candle) structure their API docs to inform best practices for Sphinx setup, especially regarding theme integration and navigation.
    PR #2: style(docs): apply and customize chosen theme

Phase 3: Automated API Documentation Generation

    Goal: Automatically generate API reference pages from Python docstrings.
    Actions:
        Configure the sphinx.ext.autodoc and sphinx.ext.napoleon (for NumPy/Google style docstrings) extensions in conf.py. Also include sphinx.ext.intersphinx for future cross-referencing.
        Create .rst files for each major module (e.g., image.rst, io.rst).
        Use autodoc directives (e.g., .. automodule::, .. autoclass::, .. autofunction::) within these files to pull in the documentation for the respective classes and functions.
        Add a table of contents to the main index.rst to link to these new API pages.
    PR #3: feat(docs): add autodoc for core api modules

Phase 4: Content Creation and Cross-Referencing

    Goal: Provide introductory guides and integrate cross-references to the Kornia core documentation.
    Actions:
        Write introductory guides covering installation, basic usage, and key differences/distinctions with Kornia core Python.
        Implement intersphinx mapping in conf.py to enable seamless cross-referencing to Kornia core docs.
        Add explicit cross-reference links (:py:obj:, :py:class:, etc.) within the kornia-rs documentation to shared objects and concepts (like ImageSize) in the Kornia core documentation.
    PR #4: docs: add introductory guides and kornia core cross-references

Phase 5: Build Automation and CI/CD Integration

    Goal: Integrate the documentation build process into the existing CI/CD pipeline and configure deployment.
    Actions:
        Create a GitHub Actions workflow that:
            Installs documentation dependencies.
            Builds the documentation using sphinx-build on every push to the main branch and on every PR.
            Crucially: Configure the workflow to deploy the generated HTML to either GitHub Pages or ReadTheDocs. (Decision on hosting strategy needs team input.)
        Update pyproject.toml or relevant project metadata to point the documentation URL to the new site once deployed.
    PR #5: ci(docs): automate build and deployment

Collaboration and Communication:

    I will share regular progress updates via GitHub issues and PRs.
    I will request reviews and guidance from @edgarriba, @cjpurackal, and any interested team members.
    Any architectural decisions or scope changes will be discussed prior to implementation.

Request for Approval and Guidance:

Before commencing Phase 1, I would appreciate confirmation on the following:

    Does this phased approach align with the team’s vision for kornia-rs documentation?
    Guidance on the preferred hosting strategy: ReadTheDocs vs. GitHub Pages.
    Are there any existing documentation standards or tooling the team prefers to reuse?
    Who would be the primary point of contact for quick reviews and clarifications?

I believe this phased approach will allow for clear progress and easy review. I'm ready to start on Phase 1 as soon as this plan is approved.