kornia/kornia-rs #341: Make immutable pyclasses in `kornia-py` frozen

ngoldbaum, 2025-04-04:
See https://github.com/kornia/kornia-rs/pull/334. 

This should be a (probably small) optimization, since there’s no need to do runtime atomic reference counting on immutable types.

ngoldbaum, 2025-04-04:
I think there's still some work:

```
goldbaum at Nathans-MBP in ~/Documents/kornia-rs on main
± rg pyclass
kornia-py/src/image.rs
54:#[pyclass(name = "ImageSize")]

kornia-py/src/icp.rs
9:#[pyclass(name = "ICPConvergenceCriteria")]
48:#[pyclass(name = "ICPResult")]

kornia-py/src/io/jpeg.rs
6:#[pyclass(name = "ImageDecoder", frozen)]
35:#[pyclass(name = "ImageEncoder", frozen)]
```

Aren't `ImageSize`, `ICPConvergenceCriteria`, and `ICPResult` all immutable?

edgarriba, 2025-04-05:
you are correct

ngoldbaum, 2025-04-07:
Can you reopen the issue?

PR #349 (2025-04-10): Add frozen attribute to Python wrapper classes
Fixes #341
closed 2025-04-10 by edgarriba