kornia/kornia-rs #351: add `draw_polygon` and `draw_filled_polygon` functions.

TomatoChocolate12, 2025-04-09:
I will work on this. Opening an issue just because I had two questions:

- should i add these functions to `draw.rs` or should they be somewhere else? maybe in the `kornia-imgproc::color` crate.
- i intend to use the *scanline-fill* algorithm to make the filled polygons. would you suggest something else?

edgarriba, 2025-04-09:
the `draw.rs` should be the right place

Mojam847, 2025-04-09:
HI @edgarriba I was thinking after draw_line, we could also add other basic drawing functions in draw.rs like:

`draw_rect, draw_filled_rect`
`draw_circle, draw_filled_circle`
`draw_ellipse, draw_filled_ellipse`
what do you think?

TomatoChocolate12, 2025-04-09:
Drawing a finite-sided polygon like a rectangle is fairly simple and should be achieved by just providing it's vertices as argument.

But, drawing an ellipse or a circle would be far more complicated as I assume there would be an irrational number of pixels if we were to try to draw it perfectly. So we might have to make some estimations and assumptions to design such functions.

 @Mojam847 if you have any ideas on how to handle shapes like ellipses and circles, I would love to discuss with you and then try to implement them!

edgarriba, 2025-04-10:
all sounds good, please make sure one function per PR choosing a good reference implementation to compare with