kornia/kornia-rs #360: New release?

ngoldbaum, 2025-05-01:
Can we get a new release on PyPI? Or is packaging gstreamer blocking that?

edgarriba, 2025-05-02:
Not blocking, sure let’s target. It’s rust and python release during the upcoming week.

ngoldbaum, 2025-05-07:
I think that's just GitHub telling you about warnings it found during the build. 

See https://github.blog/changelog/2022-10-10-github-actions-deprecating-save-state-and-set-output-commands/ for the `set-output` deprecation stuff.

The one about `pip` it looks like is being triggered by a warning in `setup-python`. I think that would need to be fixed upstream...

zboszor, 2025-05-19:
There are two problems with the source distribution on pypi.org.

- pyproject.toml does not indicate Python 3.13 support, despite #184 merged into 0.1.8 (small problem)
- kornia-py/src directory is missing so building from source fails (big problem)

As a stopgap in my https://github.com/zboszor/meta-python-ai Yocto layer, I added the contents as a separate tarball that's symlinked in for the build, which then succeeds. See https://github.com/zboszor/meta-python-ai/commit/f6da9b3e6f08680f67d1e1a692ce391813103e9f