kornia/kornia-rs #371: No Python 3.13 compatible releases on conda-forge

chippey, 2025-05-16:
I'm attempting to install a Python library that's dependent on kornia and kornia-rs via conda in a conda env that uses Python 3.13, however It's blocking as there's no candidate that resolves with Python 3.13. (And indeed looks like 3.12 is the newest available:)

```
conda search kornia-rs
Loading channels: done
# Name                       Version           Build  Channel             
kornia-rs                      0.1.3 py310h1385c18_2  conda-forge         
kornia-rs                      0.1.3 py310ha39ebd3_0  conda-forge         
<snipped  to shorten block of text, just previous releases>
kornia-rs                      0.1.8  py39h112106a_0  conda-forge         
kornia-rs                      0.1.8  py39h112106a_1  conda-forge         
kornia-rs                      0.1.9 py310hc531cc6_0  conda-forge         
kornia-rs                      0.1.9 py311hf2b9cd9_0  conda-forge         
kornia-rs                      0.1.9 py312he9d9d3d_0  conda-forge         
kornia-rs                      0.1.9  py39h112106a_0  conda-forge 
```

However, from reading the kornia docs, it looks like Python 3.13 is supported.
I don't see who's responsible for the conda packaging (from https://anaconda.org/conda-forge/kornia-rs), so I don't know who to ping about including a Py3.13 package.  If it's someone on this github repo, can 3.13 packages be added?

edgarriba, 2025-05-16:
possibly open a ticket here https://github.com/conda-forge/kornia-rs-feedstock

I see my name is there as maintainer but TBH i have no idea how the whole conda ecosystem for packaging works.

chippey, 2025-05-16:
Thanks!  I'll close this here and open a ticket over there.

closed 2025-05-16 by chippey