kornia/kornia-rs #409: Implement Zero-copy Image conversion in kornia-py

AS1100K, 2025-06-20:
Since #373 has been merged, which has allowed the use of custom allocators in the image, it is now possible to implement zero-copy immutable and mutable conversion from numpy to kornia-image. The following resouces might be helpful:

- https://github.com/PyO3/rust-numpy/discussions/432

I have also worked on it, before the PR was merged but that approach is just a prototype and IIRC there were some python interpreter crashes in some edge cases. [See Code](https://github.com/as1100k-forks/kornia-rs/commit/8657f3c12c7c24bccb7e663ca2e7af49b5bd537c#diff-9e75e791f62ce64de35e7267d9f64acc1fd26cc84dda24bb86ffd73566c74632R12)

I would deal with this, once I find time for the same

CC: @edgarriba

AS1100K, 2025-07-01:
I haven't started working on it, you can go. Since #418 is merged, you should now use arrow, maybe?

Before going ahead make sure you understand how arrow/numpy and rust store the array, and you are able to write safe zero-copy implementation as this issue requires understanding the memory fundamentally and how kornia stores data.

SkaarFacee, 2025-07-03:
Okay Sure, I can look into this given some time. Thanks for the pointers and context—I'll make sure to explore how arrow, numpy, and Rust handle memory layout.

edgarriba, 2025-12-28:
explore using dlpack in the future with the new Image class

closed 2025-12-28 by edgarriba