kornia/kornia-rs #450: Configurable max hamming distance for QuickDecode

rossng, 2025-08-10:
Very excited to see @AS1100K has been working on a Rust-native AprilTag detector.

It would be really nice if it were possible to configure the maximum hamming distance of the lookup table that is precomputed for the decoder. For some of the larger families it gets quite huge even at a distance of 2.

rossng, 2025-12-24:
> while keeping the default behavior unchanged for backward compatibility

I assume the crate isn't widely used at this point, so I would support breaking backwards compatibility to get a clean interface (unless there is a kornia-level policy that I'm unaware of)

sidd-27, 2025-12-31:
Additionally, another minor point: Is the rotation field in the quickdecode table effectively useless? It's never set anywhere and is immediately overwritten in the quick_decode_codeword function. pretty sure we can safely remove it; it would not save space in the current implementation due to padding, but still a good thing to clean up